bluekitchen/btstack #696: Fix USB level command format

ua1arn, 2025-06-24:
BLUETOOTH CORE SPECIFICATION (Version 6.1 | Vol 4, Part B) - page 1744

_For a single function Controller, the Host should address HCI command packets to the
Device. HCI command packets should be sent with the following parameters:
bmRequestType = 0x20 (Host-to-device class request, device as target)
bRequest = 0x00
wValue = 0x00
wIndex = 0x00_

Now, stm32 usbh interface use targeting as             
`phost->Control.setup.b.bmRequestType = USB_H2D | USB_REQ_RECIPIENT_INTERFACE | USB_REQ_TYPE_CLASS;`

mringwal, 2025-06-24:
Thanks for pointing this out. It's fixed on the develop branch.

Did you run into a problem with an USB Bluetooth Dongle that didn't respond to HCI Commands?

closed 2025-06-24 by mringwal

ua1arn, 2025-06-24:
You use Target "CLASS' instead of 'DEVICE'  You sure?

ua1arn, 2025-06-24:
> Did you run into a problem with an USB Bluetooth Dongle that didn't respond to HCI Commands

I am deep in initialization sequence of strange USB dongle (vid/pid same as **_TP-Link UB500 V2_**, but report USB 1.1 instead USB 2.0 - this one hung on Reset step (no solutions, simple wait timeout or skip Reset).
My sandbox - own BT host implementation for tinyusb . BT core specs say nothing about Reset (or I am blind).

ua1arn, 2025-06-24:
Can I propose 
hci_transport_h2_tinyusb.c and hci_transport_h2_tinyusb.h ?
Unfortunately, SCO packets require isochronous endpoints - not supported by _tinyusb_ OHCI/EHCI host at all... This part only stubs in my interfaces...

mringwal, 2025-06-25:
> You use Target "CLASS' instead of 'DEVICE' You sure?

Thanks for spotting this. It's fixed now. I also changed the winusb implementation for Windows.
However, just going through the Bluetooth Specification, to be correct, it should dynamically changed depending on the type of the Bluetooth Controller: single function controller or composite device. So, one is as good as the other.

mringwal, 2025-06-25:
> Can I propose hci_transport_h2_tinyusb.c and hci_transport_h2_tinyusb.h ? Unfortunately, SCO packets require isochronous endpoints - not supported by _tinyusb_ OHCI/EHCI host at all... This part only stubs in my interfaces...

I'd love a transport driver for TinyUSB (even without support for SCO). In fact, I've started to work on the USB Host part but didn't get far (due to priorities. I can enumerate interfaces and endpoints though): https://github.com/bluekitchen/hci-usb-to-uart

commit 743f5cea (2025-06-24): stm32-f4discovery-usb: fix bmRequestType for HCI Commands

Fixes #696
commit e9a8d5ae (2025-06-24): libusb: fix bmRequestType for HCI Commands

Fixes #696