bluekitchen/btstack #700: HFP AG Demo Won't Connect

tpitman, 2025-08-01:
**Describe the bug**

I have an DFRobot Firebeetle 2 WROOM-E based esp32 dev board.

I have configured and built the hfp_ag_demo for it. I am able to download it and when I try to pair/connect on my Android, iPhone or Mac I get similar results.

On Mac and iPhone I just get an error that says it isn't compatible.

On Android it asks me if I want to pair, I answer to pair, it pairs and tries to connect, but then disconnects. If I then hit the Connect button, it tries and then goes back to disconnected without any errors.

We can focus on Android because that can be easier to diagnose than Apple products.

**To Reproduce**

Steps to reproduce the behavior:
1. Run example hfp_ag_demo
2. Add a new bluetooth device on Android
3. See the device and pair with it
4. Notice after a few seconds it disconnects
5. Try to connect again and notice after a few seconds it disconnects again

**Expected behavior**

I would expect it to connect and provide the audio playback samples or sine wave depending on which I chose.

**HCI Packet Logs**

[btstack_hci.pklg.zip](https://github.com/user-attachments/files/21554195/btstack_hci.pklg.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller ESP32 WROOM-E
 - Remote device: Android 13 Phone (Pixel 4)

**Additional context**
I looked at the LogCat logs from the Android phone and saw that it seems to be trying to talk to it via BLE. I see some references to UUID values and such. I then later see these lines:

```
08-01 13:57:03.497  2827  3489 W BASM_Disconnected: DISCONNECT ignored: 08:A6:F7:63:8F:46
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=A2DP, connectionPolicy=-1
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=HEADSET, connectionPolicy=-1
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=HID_HOST, connectionPolicy=-1
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=PAN, connectionPolicy=-1
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=HEARING_AID, connectionPolicy=-1
08-01 13:57:03.497  2827  3069 V BluetoothDatabase: getProfileConnectionPolicy: device XX:XX:XX:63:8F:46 profile=BATTERY, connectionPolicy=-1
```

Which implies it does have the profiles that it needs to allow audio.

mringwal, 2025-08-02:
This is correct as a HFP AG (Audio Gateway) won't talk to an HFP AG. Your phone can only connect to a HFP HF (Hands-free) device. Please run the hfp_hf_demo if you want to connect from your smartphone.

closed 2025-08-02 by mringwal

tpitman, 2025-08-04:
> This is correct as a HFP AG (Audio Gateway) won't talk to an HFP AG. Your phone can only connect to a HFP HF (Hands-free) device. Please run the hfp_hf_demo if you want to connect from your smartphone.

Thank you for the clarification. I am new to this stuff.

Your repo is amazing.

I switched to using the handsfree demo. It works great with Android. I can't see the device on iOS, however. Is there something I need to set? I searched the code for "apple" and found a few things, but couldn't see anything that I could change.

tpitman, 2025-08-05:
> HFP HF Demo also works with iOS as is - I have a iPhone 12 Mini with the latest OS version.

I am very glad to hear that as it gives me hope.

Do you have any idea why my HF device is showing up on Android and not on iOS? I have tried 2 different iOS devices.

I have made sure that I unpaired it from the Android phone just in case that was the issue.

Are there any settings that might affect it working on iOS?

mringwal, 2025-08-05:
Try more iPhones... maybe your iPhone 15 is special, try reboot, etc..