bluekitchen/btstack #702: A2DP Demo on ESP32 WROOM-E Fire Beetle 2 Doesn't Pair on iOS or Android

tpitman, 2025-08-11:
**Describe the bug**

I did the setup for the esp32.

I open the a2dp_sink_demo project in the ports/esp32/examples folder. I build it without issues and install it on my Fire Beetle 2 dev board.

When I run it my iPhone acts like it is connecting, but never asks to pair with the device. Eventually it times out.

On Android the phone asks if I want to pair and I select "pair". It acts like it has paired, but doesn't connect and won't connect.

I have another project that is the hfp demo and it works great. When I connect using the iPhone a dialog appears asking me if I want to pair and when I select to pair it works. Same on Android.

**To Reproduce**

Steps to reproduce the behavior:
1. Run example a2dp_sink_demo
2. Connect to remove device iPhone 12
3. Note that the iPhone never asks to pair
4. Note that eventually it times out

**Expected behavior**

I would expect it to pair and connect and then be able to receive audio from my iPhone.

**HCI Packet Logs**

[packet_log.pklg.zip](https://github.com/user-attachments/files/21722432/packet_log.pklg.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: develop
 - Bluetooth Controller DFRobot Fire Beetle 2 esp32-wroom-e
 - Remote device: iPhone 12 with iOS 18.6

mringwal, 2025-08-11:
Hi. There's just a disconnect in the log shortly after the connect. Maybe it isn't your phone after all, but the ESP32 dev kit - or you're living in a very challenging RF environment. 

Could you try a different ESP32 dev kit? (Sorry, no better idea right now).

mringwal, 2025-08-16:
HSP/HFP actually requires working SCO connections in addition to ACL connections, so in general, if HFP works, A2DP should work as well. In your log, the connection is lazily closed by the remote device after doing a single SDP query. The query is for all L2CAP services, which is basically everything. In our a2dp_sink_demo, there should be 2 SPD Records, one for A2DP/AVDTP and one for AVRCP/AVCTP. Did you make any changes to our demo? Please try to run the a2dp_sink_demo without changes on your ESP32 Wroom board.

winzkigermany, 2025-08-28:
Seeing the same here. Pairing is working just fine.
But connecting A2DP does not work, because the connecting client never receives an SDP answer, that announces A2DP capabilities of the esp32.
The SDP queries (from my pixel9 to the esp32) don't receive an answer (or a complete answer) from the esp32, although the SDP datasets are configured correctly on the btstack side.

@mringwal: Maybe my dump from our last conversation helps you to pinpoint the problem...
Last time, I was able to somehow make it work by erasing all flash of esp32, but not any more now. I'm stuck with the exact same problem again (develop branch).

mringwal, 2025-08-28:
@winzkigermany Thanks for making the connection. Do you have older binaries of your software you can test against to see if there was a change in BTstack and/or ESP-IDF?

The strange thing is that there is no problem between iPhone and BTstack with an USB dongle. Also, running A2DP Source Demo of BTstack on USB dongle works also fine with the same ESP32 that failed with the iPhone.

I'll get some HCI traces from the iPhone later and look for differences between ESP32 and USB dongle. In the local logs, there is no error, it's just that iOS does not send the next SDP query. Let's see if the iPhone did actually receive the SDP response.

tpitman, 2025-08-28:
> [@tpitman](https://github.com/tpitman) : What esp-idf version are you working on? My project is currently on esp-idf 5.4.0

I am pretty sure when I started all of this I was using 4.4.x. I just tried 5.5.0 and it does the same thing.

mringwal, 2025-08-29:
As mentioned, the same board works now, even in the same location I've tested the first time without reflashing it. 

To move forward, we need more sample data: could you test both HFP HF as well as A2DP Sink on different ESP32 boards with different phones/laptops to see if there is some pattern? The HCI Traces from the ESP32 show no error.

@tpitman Could you get an HCI trace from your Android and/or iPhone? It's easy on iOS following this instructions: https://developer.apple.com/bug-reporting/profiles-and-logs/ (although you probably need a (free) iOS develop account). I was able to get logs from Android the last time directly with Wireshark although it depends on your Android version. Ideal would be logs from ESP32 as well as from the remote that cover the issue.

mringwal, 2025-08-29:
@tpitman thanks for bringing this up anyways. Feel free to contact us directly for commercial projects.

tpitman, 2025-08-29:
Will do.

winzkigermany, 2025-09-03:
I am currently going up and down with btstack versions and the esp-idf bt component versions. Nothing changes for the better.
Pairing works fine. But then when I click "connect" on my phone (because the esp32 is showing as paired, but not connected)
then my esp32 does not show /any/ incoming packets.
I'll need to try to get back to some working state, before I can go on juggling with versions.